AdnanHodzic/displaylink-debian #827: macbook pro  2016 touchbar ubuntu 23.04

ozpakh, 2023-04-28:
macbook pro  2016 touchbar ubuntu 23.04
merci

ozpakh, 2023-05-02:
macbook pro 2016 touchbar ubuntu 23.04
non affiche pas 
avec la commander 
echo -e "\n# applespi\napplespi\nspi_pxa2xx_platform\nintel_lpss_pci" >> /etc/initramfs-tools/modules

apt install dkms
git clone https://github.com/PatrickVerner/macbook12-spi-driver.git /usr/src/applespi-0.1
dkms install -m applespi -v 0.1
merci d'avancez

Barabazs, 2023-06-04:
Please create a new issue with all the debug information and a proper description of the issue.

closed 2023-06-04 by Barabazs

ozpakh, 2023-06-04:
mbk 2016 ubuntu 23.04 
 
make -j8 KERNELRELEASE=6.2.0-20-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for applespi: 0.1 not found
Error! Bad return status for module build on kernel: 6.2.0-20-generic (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information