AdnanHodzic/displaylink-debian #828: Need support for Kali Linux 2023.1

m-attique, 2023-04-29:
Hi,

while installing on Kali Linux 2023.1, I'm getting following error:

```
update-alternatives: using /usr/bin/bison.yacc to provide /usr/bin/yacc (yacc) in auto mode
Setting up evdi (1.13.1-18) ...
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Error! DKMS tree already contains: evdi-1.13.1
You cannot add the same module/version combo more than once.
dpkg: error processing package evdi (--configure):
 installed evdi package post-installation script subprocess returned error exit status 3
Setting up flex (2.6.4-8.1) ...
Setting up linux-source (6.1.20-2kali1) ...
Setting up libfl-dev:amd64 (2.6.4-8.1) ...
dpkg: dependency problems prevent configuration of displaylink-driver:
 displaylink-driver depends on evdi (>= 1.13.1); however:
  Package evdi is not configured yet.
 displaylink-driver depends on evdi (<< 1.14.0); however:
  Package evdi is not configured yet.

dpkg: error processing package displaylink-driver (--configure):
 dependency problems - leaving unconfigured
Processing triggers for doc-base (0.11.1) ...
Processing 1 added doc-base file...
Processing triggers for libc-bin (2.36-9) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for kali-menu (2023.2.0) ...
Errors were encountered while processing:
 evdi
 displaylink-driver
E: Sub-process /usr/bin/dpkg returned an error code (1)
linux-source installation failed.  Aborting.
```

**Below is the output of the Debug mode of Script:**

```
# ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 6.1.0-kali7-amd64

---------------- DisplayLink info ----------------

Driver version: 1.13.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation HD Graphics 530 (rev 06)
VGA (3D): NVIDIA Corporation GM107M [GeForce GTX 950M] (rev a2)
X11 version: 21.1.7-2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x43 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x23e cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Please guide me in correct direction if I'm missing something and help me install DisplayLink Docking Station on my Kali Laptop.

Kind Regards,
Attique

Barabazs, 2023-06-04:
It seems like evdi was already installed. Please make sure it is completely uninstalled before retrying the install procedure.
Also copy the entire terminal output from the install procedure.

mzhk, 2023-06-07:
You need a new evdi version for kernel 6.* (and not the one that's part of the package that is downloaded by the installer from DisplayLink). Currently, requires some manual tinkering as described here:
https://code.berrydejager.com/Fix-DisplayLink_drivers-linux-kernel-6/

SteveMKII, 2023-06-18:
I ran into a similar issue when I upgraded from Debian 11 to Debian 12, the driver installed fine but I got this error `Driver version: 1.11.0 1.13.1 DisplayLink service status:  EVDI service version: /sys/devices/evdi/version not found `

What solved it for me was `sudo apt install evdi-dkms`, once I did that the monitor worked instantly. Not sure if it will fix your issue but hopefully it helps

Kaldor37, 2023-06-21:
> I ran into a similar issue when I upgraded from Debian 11 to Debian 12, the driver installed fine but I got this error `Driver version: 1.11.0 1.13.1 DisplayLink service status: EVDI service [...]

Thanks for the comment Steve, I had exactly the same issue on my new build with Debian 12. Everything installed seamlessly, all but evdi which was missing. Installing **evdi-dkms** made it work without rebooting and now it's working really fine.

Thanks @AdnanHodzic and the other participants for all the work it's a really nice project, very helpful for using Debian on a laptop with today's technologies.

m-attique, 2023-06-21:
Thank you all guys... 

Issue has finally resolved for me by, as mentioned by @Kaldor37 & @SteveMKII, running the `sudo apt install evdi-dkms` command. The same tool installation resolved the issue for me as well.

@mzhk, the same command install its latest version 6.1.X.

KR :+1:

closed 2023-06-21 by m-attique