AdnanHodzic/displaylink-debian #829: No providers are showing

JTurner7, 2023-05-05:
```bash
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: jammy
Kernel: 5.19.0-41-generic

---------------- DisplayLink info ----------------

Driver version: 1.13.1
DisplayLink service status: up and running
EVDI service version: 1.13.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [1849:a780]
VGA: Intel Corporation Device a780 (rev 04)
VGA (3D): 
X11 version: 21.1.4-2ubuntu1.7~22.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```
Ubuntu 22.04.2
Using a Teolant 11 in 1 usb docking station to display on two monitors.
I tried deleting the 20-displaylink.conf and uninstalling then reinstalling it, but my Providers is zero.
Can I get some help with this issue?

Barabazs, 2023-06-04:
Can you try replacing your current config with this:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

JTurner7, 2023-06-04:
I replaced it then uninstalled, but it said I didn't have the .patch file. Then I reinstalled and the config didn't appear, so I pulled the repo again and uninstalled, and now it can't find the /opt/displaylink/displaylink-installer.sh. Now there is no 20-displaylink.conf because it is installing improperly.

uninstall
```bash
displaylink-debian ±|master|→ sudo ./displaylink-debian.sh --uninstall
[sudo] password for johnathan: 

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.19.0-42-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

bash: /opt/displaylink/displaylink-installer.sh: No such file or directory

-------------------------------------------------------------------

Performing clean-up
```
install
```bash
displaylink-debian ±|master|→ sudo ./displaylink-debian.sh --install

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.19.0-42-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/5416?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2023-06-04 09:00:41--  https://www.synaptics.com/sites/default/files/exe_files/2023-04/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.7-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 54.245.106.105
Connecting to www.synaptics.com (www.synaptics.com)|54.245.106.105|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17130641 (16M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.7.zip’

DisplayLink_Ubuntu_5.7.zip                         100%[================================================================================================================>]  16.34M  11.8MB/s    in 1.4s    

2023-06-04 09:00:43 (11.8 MB/s) - ‘DisplayLink_Ubuntu_5.7.zip’ saved [17130641/17130641]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.7.zip
  inflating: 5.7/DisplayLink USB Graphics Software for Ubuntu 5.7.0-61.129-Release Notes.txt  
  inflating: 5.7/displaylink-driver-5.7.0-61.129.run  
Creating directory displaylink-driver-5.7.0-61.129
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.7.0-61.129  100%  
patching file 5.7/displaylink-driver-5.7/displaylink-installer.sh
patch unexpectedly ends in middle of line

Installing driver version: 5.7

Distribution discovered: Ubuntu 22.04.2 LTS
WARNING: Version 1.13.1 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory
```

JTurner7, 2023-06-04:
Okay I realized after commenting that I have to reboot after uninstall, so I did that and the install worked!
I then changed the 20-displaylink.conf to what you recommended. After plugging it in one of the hdmi ports on the device started working, but the other one doesn't (strange). I still don't have any providers though.
debug
```bash
--------------- Linux system info ----------------

Distro: Ubuntu
Release: jammy
Kernel: 5.19.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 1.13.1
DisplayLink service status: up and running
EVDI service version: 1.13.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [1849:a780]
VGA: Intel Corporation Device a780 (rev 04)
VGA (3D): 
X11 version: 21.1.4-2ubuntu1.7~22.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```