AdnanHodzic/displaylink-debian #830: debian 11 update kills display link

vallezwh, 2023-05-10:
successfully used displaylink with debian 11 for a year

after kernal update no longer works 

did a reinstall with latest version of desplaylink-debian

tried different versions of 20-displaylink.conf including
blank

have used displaylink xorg patch from synaptics before, but 
not making any difference with or without

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-22-amd64

---------------- DisplayLink info ----------------

Driver version: 1.13.1
1.9.0+dfsg
DisplayLink service status: 
EVDI service version: 1.9.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 9a78 (rev 01)
VGA (3D): 
X11 version: 1.20.11-1+deb11u6

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x4a cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 7 associated providers: 0 name:modesetting
Provider 1: id: 0x802 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xd9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

----------------------------------------------------------------------------------sudo systemctl status displaylink-driver.service
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/lib/systemd/system/displaylink-driver.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Wed 2023-05-10 07:53:02 MDT; 17min ago
    Process: 755 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install $(ls -t /usr/src | grep evdi | >
    Process: 757 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=0/SUCCESS)
   Main PID: 757 (code=exited, status=0/SUCCESS)
        CPU: 2.277s

May 10 07:52:07 chama systemd[1]: Starting DisplayLink Driver Service...
May 10 07:52:07 chama systemd[1]: Started DisplayLink Driver Service.
May 10 07:53:01 chama systemd[1]: Stopping DisplayLink Driver Service...
May 10 07:53:02 chama systemd[1]: displaylink-driver.service: Succeeded.
May 10 07:53:02 chama systemd[1]: Stopped DisplayLink Driver Service.
May 10 07:53:02 chama systemd[1]: displaylink-driver.service: Consumed 2.277s CPU time.

--------------------------------------------------------------------------------------------------------------------------------------------
journalctl -u

-- Boot 4823e03739d344438fa50bec54bc7867 --
May 10 07:52:07 chama systemd[1]: Starting DisplayLink Driver Service...
May 10 07:52:07 chama systemd[1]: Started DisplayLink Driver Service.
May 10 07:53:01 chama systemd[1]: Stopping DisplayLink Driver Service...
May 10 07:53:02 chama systemd[1]: displaylink-driver.service: Succeeded.
May 10 07:53:02 chama systemd[1]: Stopped DisplayLink Driver Service.
May 10 07:53:02 chama systemd[1]: displaylink-driver.service: Consumed 2.277s CPU time.

---------------------------------------------------------------

vallezwh, 2023-05-11:
Uninstalled displaylink, reboot, reinstall linux headers, displaylink-debian install still fails with same error

tshelley, 2023-05-12:
#821 Same issue.

closed 2023-05-12 by vallezwh