AdnanHodzic/displaylink-debian #832: Ubuntu 22.04 with nvida gtx 1650 do not show detected monitor at gnome display configuration

GleidsonLeite, 2023-05-11:
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: jammy
Kernel: 5.19.0-41-generic

---------------- DisplayLink info ----------------

Driver version: 1.13.1
DisplayLink service status: up and running
EVDI service version: 1.13.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [UHD
GTX
VGA: Intel Corporation CometLake-H GT2 [UHD Graphics] (rev 05)
NVIDIA Corporation TU117M [GeForce GTX 1650 Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 21.1.4-2ubuntu1.7~22.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 5 name:modesetting
Provider 1: id: 0x2a5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x284 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x263 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x242 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1ff cap: 0x2, Sink Output crtcs: 4 outputs: 1 associated providers: 1 name:NVIDIA-G0

-------------------------------------------------------------------

IsaqueAmorim, 2024-04-02:
Estou Enfrentando o mesmo problema

Barabazs, 2024-05-03:
The issue you're describing is not related to `displaylink-debian`.
I would suggest asking on the displaylink forums or on the appropriate stackoverflow.

closed 2024-05-03 by Barabazs