AdnanHodzic/displaylink-debian #833: Install Error! Bad return status for module build on kernel: 5.10.0-23-amd64 (x86_64)

luizgsbraz, 2023-05-17:
Found this when on installing:

```
Distribution discovered: Debian GNU/Linux 11 (bullseye)

Installing
[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]

Creating symlink /var/lib/dkms/evdi/1.13.1/source ->
                 /usr/src/evdi-1.13.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.10.0-23-amd64 all INCLUDEDIR=/lib/modules/5.10.0-23-amd64/build/include KVERSION=5.10.0-23-amd64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-23-amd64 (x86_64)
Consult /var/lib/dkms/evdi/1.13.1/build/make.log for more information.

```

luizgsbraz, 2023-05-17:
```
$ cat /etc/issue
Debian GNU/Linux 11 \n \l
```

vallezwh, 2023-05-22:
see #821

iamdey, 2023-05-30:
This is probably due to an error when building evdi, see #821 
I think this should be closed as duplicate (sorry for the spam, but it seems to be duplicated several times)

Barabazs, 2023-06-04:
@luizgsbraz Please try the fix described in [Make error Bad return status for module build on kernel...](https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/common-issues.md#make-error-bad-return-status-for-module-build-on-kernel)

closed 2023-06-04 by Barabazs