AdnanHodzic/displaylink-debian #844: Fails on Ubuntu 23.04 Lunar

SimoParmeg, 2023-06-22:
I followed the instruction but I'm receiving this error. Any help will be welcome

sudo: unable to execute ./displaylink-debian.sh: No such file or directory

hh-it-co, 2023-07-01:
You need to create a 'resources'-directory in your working-directory and place this file into it. https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/resources/displaylink-installer.patch
This will resolve the 'No such file or directory'-error. But it leads into other errors since the paths inside the scripts are not correct for ubuntu 23.04. But i will do further investigation. It should be not that hard to resolve.

Kaus507, 2023-12-17:
Hello, 
I had the same error:
Kubuntu 22.04
KDE-Plasma-Version: 5.24.7
KDE-Frameworks-Version: 5.92.0
Qt-Version: 5.15.3
Kernel-Version: 6.2.0-37-generic (64-bit)
I created resources directory and saved the upper mentioned file under the name 
mentioned in the error message.
Thank you very much
Best Regards
Klaus

Barabazs, 2024-05-03:
Please clone/download the entire repo to avoid issues like this.

closed 2024-05-03 by Barabazs