AdnanHodzic/displaylink-debian #849: Display link not detecting monitor

thelovekesh, 2023-07-09:
Debug report:
```log

-------------------------------------------------------------------

Starting Debug ...

--------------- Linux system info ----------------

Distro: Pop
Release: jammy
Kernel: 6.2.6-76060206-generic

---------------- DisplayLink info ----------------

Driver version: 1.13.1
DisplayLink service status: up and running
EVDI service version: 1.13.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [103c:8a15]
[GeForce
VGA: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
NVIDIA Corporation GA107M [GeForce RTX 3050 Mobile] (rev a1)
VGA (3D): 
X11 version: 21.1.4-2ubuntu1.7~22.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b7 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x29b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x27a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x259 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x238 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1e4 cap: 0x2, Sink Output crtcs: 4 outputs: 3 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

I have tried it with and without a Dell WD19TBS docking station as well. Any help is much appreciated.

ririck0, 2023-07-18:
I have the same issue with a Dell D6000 doking station. I don't really understand what's going on, because everything seems to work without errors, but no monitor connected on displayport is detected.
`-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: trixie
Kernel: 6.3.0-1-amd64

---------------- DisplayLink info ----------------

Driver version: 1.13.1
1.14.0
DisplayLink service status: up and running
EVDI service version: 1.14.0

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [UHD
RTX
VGA: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630]
NVIDIA Corporation TU106M [GeForce RTX 2060 Mobile] (rev a1)
VGA (3D):
X11 version: 21.1.7-3
X11 configs: /etc/X11/xorg.conf /etc/X11/xinit/nvidia-xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b7 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x2cf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2ae cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x28d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x26c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1ff cap: 0x2, Sink Output crtcs: 3 outputs: 5 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
`

thelovekesh, 2023-10-01:
My dock station was missing DisplayLink support hence it was not working. Working as expected after testing with another docking station.

closed 2023-10-01 by thelovekesh