AdnanHodzic/displaylink-debian #850: Debian 12 installation and kernel upgrade compilation error

Roedy13, 2023-07-12:
Linux 6.1.0-10-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.37-1 (2023-07-03) x86_64 GNU/Linux

In the file  `/var/lib/dkms/evdi/1.13.1/source/evdi_drm.h` there is an include statement `#include "drm.h"

In debian 12, this path should be `drm/drm.h`

I  was able to trick it with a symlink

```
cd /usr/src/linux-headers-6.1.0-10-common/include/uapi
sudo ln -fs drm/drm.h drm.h
```
I assume this folder could also be added to the `-I` options when compiling

gddrew, 2023-07-20:
This appears to be related the issue I reported yesterday, would you agree? https://github.com/AdnanHodzic/displaylink-debian/issues/852#issue-1812288952

Roedy13, 2023-07-20:
> This appears to be related the issue I reported yesterday, would you agree? [#852 (comment)](https://github.com/AdnanHodzic/displaylink-debian/issues/852#issue-1812288952)

It definitely could be...  I   This is what I ended up doing.

- uninstall the display link driver
- update your kernel
- reboot, make sure the new kernel boots
- add the symlink to `drm.h` that I posted above
- reinstall display link driver

It's gonna be super annoying having to do this every time there's a kernel update.

Roedy13, 2023-07-20:
I just dug around the evdi source code and there definitely is a `-I include/uapi/drm/` in the makefile.   Also that file has always been there (at least back to kernel 4.0).  I never had a problem with this installer in Debian 11, but this popped up in Debian 12.   Now that I've actually dug into the code I have *no* idea why this didn't work out of the box, or why adding that symlink fixed anything.

gddrew, 2023-07-21:
> It's gonna be super annoying having to do this every time there's a kernel update.

Agreed. I think I'll just forgo the kernel update because it was not easy and straightforward getting this to work in the first place! My next computer is either going to have two HDMI ports, or it will be a Mac. Either way, it will be one less irritation to deal with.

Roedy13, 2023-07-27:
@gddrew For what it's worth... I found out that thunderbolt docks work out of the box on my system if that's an option for you.

gddrew, 2023-07-28:
I was wondering about that. What brand/model did you get?

Roedy13, 2023-07-28:
I got a "thinkpad workstation thunderbolt 4" dock but  I realized after the fact that the "thinkpad universal thunderbolt 4" dock is about $100 cheaper so I would have gotten that one.  

The only difference is that lenovo workstation laptops still have a dedicated DC power input so the connector cable is a USB-C bundled with the lenovo DC power cable... also the power brick is ENORMOUS.

Kaldor37, 2023-08-08:
Many thanks

> It definitely could be... I This is what I ended up doing.
>
>     * uninstall the display link driver
>
>     * update your kernel
>
>     * reboot, make sure the new kernel boots
>
>     * add the symlink to [...]

Thank you very much @Roedy13 for this workaround, I was experiencing the same problem since the last upgrade and it helped me solve it.

@AdnanHodzic can you think of some way to implement a fix for this ?

Have a nice day you all.

Roedy13, 2023-08-09:
Yeah this is goofy because the header file structure doesn't change between linux 5.x and 6.x, and the `-I` includes are the same.

Maybe it's related to a newer version of gcc?  Debian 11 has gcc 10 and Debian12 has gcc 12