AdnanHodzic/displaylink-debian #859: Support for TailsOS

cyfist117, 2023-08-11:
Hi, I often find myself in need to use Tails secure OS to use the internet from where I am located. TailsOS is built on Debian 11 and I was able to get a Dell D6000 docking station but when I connect to the dock I have no monitor output or network. USB works though. I tried to use your install script but there are packages required that are not available to Tails (older versions of those packages are tho). Is there any way that you can help modify a version of your script to make it work for Tails? Thank you so much.

4SH-gaupee, 2024-04-09:
Hi, if the issue is still relevant to you, could you please try the last release ? https://github.com/AdnanHodzic/displaylink-debian/releases/tag/v5.8.0

Barabazs, 2024-05-03:
@cyfist117 
Please post the output of `sudo ./displaylink-debian.sh --debug`