AdnanHodzic/displaylink-debian #861: URL of  Post Installation Guide broken

acomjean, 2023-08-31:
Its a very minor point:  Thanks for the package. 

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] N

Please read Post Installation Guide: http://bit.ly/2TbZleK

redirects to: https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md

but that page doesn't work.  I found the guide, it should link to here:

https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/post-install-guide.md

acomjean, 2023-08-31:
And the second link is also broken.  Love the humor.

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] B

-----------------------------

Wrong value. Concentrate!

-----------------------------

devincurrie, 2023-09-13:
Same issue too.

Barabazs, 2024-02-23:
Thank you for reporting this.

PR #900 (2024-02-23): doc: update some bit.ly links
Closes #861
closed 2024-02-23 by Barabazs