AdnanHodzic/displaylink-debian #885: Support for BunsenLabs Boron / Debian 12 Bookworm

jimjamz, 2024-01-13:
```
---------------------------------------------------------------

Unsuported platform: Bunsenlabs boron
Full list of all supported platforms: http://bit.ly/2zrwz2u

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Bunsenlabs
Release: boron
Kernel: 6.1.0-16-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 21.1.7-3+deb12u4

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

lsb_release -a:
```
Distributor ID:	Bunsenlabs
Description:	BunsenLabs GNU/Linux 12 (Boron)
Release:	12
Codename:	boron
```

uname -r:
`6.1.0-16-amd64`

Pull-request to be submitted.

jimjamz, 2024-01-13:
I would like to submit a pull-request but do not have write-access to the repo to push a new feature branch.

Barabazs, 2024-01-15:
> I would like to submit a pull-request but do not have write-access to the repo to push a new feature branch.

Thank you for wanting to contribute. You should fork the repo, make the changes in your own fork and then create a PR.

jimjamz, 2024-01-25:
Thanks for the reply.  I've gone ahead and done that.  I've created a forked pull-request for this issue, which is ready for PR review: https://github.com/AdnanHodzic/displaylink-debian/pull/890

closed 2024-04-09 by Barabazs