AdnanHodzic/displaylink-debian #886: Support for Linux Mint 21.3 as third screen

h2lsoft, 2024-01-14:
Hi,

here my config :

- linux mint 23.1
- linux headers : 5.15.0-91-generic
- nvidia card GT 930 (last compatible official driver)
- screen 1: connected hdmi (master screen)
- screen 2: dvi (second screen)
- screen 3: displaylink connected

If I use driver xgeneric-nouveau, it works displaylink shows screens but with bad resolution.
If I install displaylink-debian => I have only one screen that works, display-driver 5.8 donwloaded from script.

Thanks

TeddybearCrisis, 2024-02-27:
Mostly FYI:
Without really taking my time to understand what [displaylink-debian.sh](https://github.com/AdnanHodzic/displaylink-debian/blob/master/displaylink-debian.sh) does I edited the script locally. It now executes the install when the codename is "virginia". However this did **not work** in terms of a leading to the third monitor to be detected. So it's not simply the script blocking the install. There's proper adaptation work required.

Thanks in advance to the person solving this.

leonp5, 2024-04-02:
[jskalski2024](https://github.com/jskalski2024) wrote in his pull request, where he adds LM 21.3 support, that he had to delete `20-displaylink.conf` too.
Here you can see it: https://github.com/AdnanHodzic/displaylink-debian/pull/906

closed 2024-05-03 by Barabazs