AdnanHodzic/displaylink-debian #889: DISPLAYLINK is not working on kali kernel 6.5.0-kali3

diegopoueriet, 2024-01-25:
Hi all, 

I install a new computer with Kali 2023.4 kernel: 6.5.0-kali3-amd64 and the installations looks like works but its not working. 

Some one have any update of displaylink to fix this?

pedroflor, 2024-01-29:
Hi @diegopoueriet 
I just re-installed `Kali` + `displaylink-debian` and everything works flawlessly.

Here is some data of my system.

```bash
lsb_release -a
No LSB modules are available.
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	2023.4
Codename:	kali-rolling
```

```bash
uname  -r
6.5.0-kali3-amd64
```

Could you provide more technical information?
Log files, specific errors, etc.

diegopoueriet, 2024-01-29:
Thanks for your answer, after my post, I did some research and by modifying the /etc/X11/xorg.conf.d/20-displaylink.conf  file I was able to get results. 

Just in case if some one else need to know, I just coment the original file and put this follow lines. 

#Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection
#
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

THANKS!!