AdnanHodzic/displaylink-debian #896: Display link service failed Kali 6.6.9-amd64

Mathis-Dory, 2024-02-15:
Hey,
I did a fresh install of Kali (see output below) and for some reasons display link service failed. I already tried to edit / remove the /etc/X11/xorg.conf.d/20-displaylink.conf file without success and even worst, after trying to reinstall it, it corrupted my boot loader and I had to reinstall Kali again and again...

`
--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 6.6.9-amd64

---------------- DisplayLink info ----------------

Driver version: 1.14.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
nouveau
Subsystem: [UHD
RTX
VGA: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630]
NVIDIA Corporation TU106M [GeForce RTX 2060 Mobile] (rev a1)
VGA (3D): 
X11 version: 21.1.11-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 1 name:modesetting
Provider 1: id: 0xfd cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 2 associated providers: 1 name:modesetting

-------------------------------------------------------------------
`

ankittiwari2001, 2024-03-29:
got any progress ? i'm also facing same problem

Mathis-Dory, 2024-04-01:
> got any progress ? i'm also facing same problem

Nop, I finally decided to move on Ubuntu.