AdnanHodzic/displaylink-debian #899: displaylink issue for Pop!_OS 22.04 LTS

bdawg89, 2024-02-22:
Hi I am unable to install displaying Debian on pop os 22.04. Below is the debug information. It seems related to evdi and it not being installed properly?  I did a full refresh of my box and tried to install it but I am not getting anywhere any help would be appreciated.  

 
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Pop
Release: jammy
Kernel: 6.6.10-76060610-generic

---------------- DisplayLink info ----------------

Driver version: 1.14.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Inc.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne (rev d1)
VGA (3D): 
X11 version: 21.1.4-2ubuntu1.7~22.04.8
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x56 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:AMD Radeon Graphics @ pci:0000:03:00.0

ankittiwari2001, 2024-03-29:
have you got any progress regarding this issue ? i'm facing the same

Barabazs, 2024-05-02:
Please post the output of the installation procedure.