AdnanHodzic/displaylink-debian #908: Notebook won't boot anymore after executing ./displaylink-debian.sh

ReneNyffenegger, 2024-03-31:
When using wget to download `displaylink-debian.sh` and executing this script, I get an error message about a missing patch file (see https://github.com/AdnanHodzic/displaylink-debian/issues/907).
I don't encounter this error when cloning the github repository and executing the script in the repository. However, with this approach, the notebook won't boot anymore.
So, is there a possibility to revert the changes of the script?

ReneNyffenegger, 2024-04-01:
After removing `/etc/X11/xorg.conf.d/20-displaylink.conf` in recovery mode, the notebook does boot again. Of course, I am not sure if the install script messed up other parts of the OS as well.

leonp5, 2024-04-05:
If you run the script it asks you, if you want to install or uninstall the driver. This

648trindade, 2024-04-19:
I'm having the exact same problem. After installing and rebooting, the computer get a black screen.

My laptop has hybrid graphics, with a integrated Radeon GPU and a dedicated NVIDIA GTX 1650, not sure if this is the problem.

It looks like it is working, although I haven't tested yet. I'm getting this message `DisplayLink service status: up and running`

Here is the output of my `--debug` command:

```
--------------- Linux system info ----------------

Distro: Debian
Release: trixie
Kernel: 6.6.15-amd64

---------------- DisplayLink info ----------------

Driver version: 1.14.1
1.14.2
1.14.2+dfsg
DisplayLink service status: up and running
EVDI service version: 1.14.2

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Vega
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev c5)
VGA (3D): NVIDIA Corporation TU117M [GeForce GTX 1650 Mobile / Max-Q] (rev a1)
X11 version: 21.1.11-2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x54 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 2 associated providers: 4 name:AMD Radeon Graphics @ pci:0000:05:00.0
Provider 1: id: 0x27c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x25b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x23a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x219 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1f8 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G0

-------------------------------------------------------------------
```

output of `/etc/X11/xorg.conf.d/20-displaylink.conf`:

```
Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection
```

648trindade, 2024-04-19:
Okay, I've managed to make it boot (kudos to chat GPT). I've changed the conf file to
1. List my radeon device instead of an intel one
2. Select the radeon as the device for screen 0

```
Section "ServerLayout"
    Identifier "layout"
    Screen 0 "radeon"
    Inactive "nvidia"
EndSection

Section "Device"
    Identifier "radeon"
    Driver "amdgpu"
    Option "AccelMethod" "glamor"
EndSection

Section "Screen"
    Identifier "radeon"
    Device "radeon"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection
```

648trindade, 2024-04-20:
Okay, I've tested the adapters that I've mentioned earlier

* None of them worked with the fixed xorg conf file
* Without a xorg conf file, I've managed to make one of them to work
* The WAVLINK one works well. Here a dump of `dmesg` when I connect it to the laptop
```
[  355.034154] usb 2-1: new SuperSpeed USB device number 13 using xhci_hcd
[  355.054831] usb 2-1: New USB device found, idVendor=17e9, idProduct=4301, bcdDevice=36.05
[  355.054845] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  355.054851] usb 2-1: Product: USB3 to HDMI
[  355.054855] usb 2-1: Manufacturer: DisplayLink
[  355.054858] usb 2-1: SerialNumber: U3HD338915852
[  355.187287] usb 2-1: Warning! Unlikely big volume range (=511), cval->res is probably wrong.
[  355.187304] usb 2-1: [15] FU [USB AUDIO Playback Volume] ch = 6, val = -8176/0/16
[  355.188821] usb 2-1: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
[  355.188832] usb 2-1: [12] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
[  355.714453] evdi: [I] (card2) Opened by Task 4724 (DesktopManagerE) of process 4711 (DisplayLinkMana)
[  355.716158] evdi: [I] (card2) Added i2c adapter bus number 8
[  355.716166] evdi: [I] (card2) Connected with Task 4724 (DesktopManagerE) of process 4711 (DisplayLinkMana)
[  355.716174] evdi: [I] (card2) Connector state: connected
[  355.735393] evdi: [I] (card2) Connector state: connected
[  355.735601] evdi: [I] (card2) Edid property set
[  355.735795] evdi: [I] (card2) Connector state: connected
[  355.735986] evdi: [I] (card2) Edid property set
[  355.774524] evdi: [I] (card2) Notifying display power state: on
[  355.774546] evdi: [I] (card2) Notifying mode changed: 1366x768@60; bpp 32; pixel format XR24 little-endian (0x34325258)
[  355.774554] evdi: [I] (card2) Notifying display power state: on
[  355.831006] evdi: [W] evdi_painter_connect:886 (card2) Double connect - replacing 00000000124e4824 with 00000000124e4824
[  355.831019] evdi: [I] (card2) Connected with Task 4724 (DesktopManagerE) of process 4711 (DisplayLinkMana)
[  355.831026] evdi: [I] (card2) Connector state: connected
[  355.894294] evdi: [I] (card2) Connector state: connected
[  355.894495] evdi: [I] (card2) Edid property set
[  355.919914] evdi: [I] (card2) Notifying mode changed: 1366x768@60; bpp 32; pixel format XR24 little-endian (0x34325258)
```
* The DELL UD22 one didn't work. All functionalities of the dock works, though, with exception of the HDMI connection. Here's a dump of `dmesg` as well:
```
[  630.284620] usb 1-1: new high-speed USB device number 19 using xhci_hcd
[  630.440144] usb 1-1: New USB device found, idVendor=2109, idProduct=2822, bcdDevice=37.64
[  630.440163] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  630.440169] usb 1-1: Product: USB2.0 Hub             
[  630.440173] usb 1-1: Manufacturer: VIA Labs, Inc.         
[  630.440177] usb 1-1: SerialNumber: 000000001
[  630.501103] hub 1-1:1.0: USB hub found
[  630.501518] hub 1-1:1.0: 5 ports detected
[  630.563996] usb 2-1: new SuperSpeed USB device number 14 using xhci_hcd
[  630.716461] usb 2-1: New USB device found, idVendor=2109, idProduct=0822, bcdDevice=37.64
[  630.716475] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  630.716481] usb 2-1: Product: USB3.1 Hub             
[  630.716485] usb 2-1: Manufacturer: VIA Labs, Inc.         
[  630.716489] usb 2-1: SerialNumber: 000000001
[  630.740210] hub 2-1:1.0: USB hub found
[  630.740615] hub 2-1:1.0: 4 ports detected
[  631.310398] usb 1-1.3: new high-speed USB device number 20 using xhci_hcd
[  631.465401] usb 1-1.3: New USB device found, idVendor=2109, idProduct=2822, bcdDevice=38.23
[  631.465418] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  631.465424] usb 1-1.3: Product: USB2.0 Hub             
[  631.465429] usb 1-1.3: Manufacturer: VIA Labs, Inc.         
[  631.465432] usb 1-1.3: SerialNumber: 000000001
[  631.524985] hub 1-1.3:1.0: USB hub found
[  631.525285] hub 1-1.3:1.0: 5 ports detected
[  631.555994] usb 2-1.3: new SuperSpeed USB device number 15 using xhci_hcd
[  631.725351] usb 2-1.3: New USB device found, idVendor=2109, idProduct=0822, bcdDevice=38.23
[  631.725365] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  631.725370] usb 2-1.3: Product: USB3.1 Hub             
[  631.725374] usb 2-1.3: Manufacturer: VIA Labs, Inc.         
[  631.725378] usb 2-1.3: SerialNumber: 000000001
[  631.748308] hub 2-1.3:1.0: USB hub found
[  631.748476] hub 2-1.3:1.0: 4 ports detected
[  631.859054] usb 1-1.4: new high-speed USB device number 21 using xhci_hcd
[  632.014918] usb 1-1.4: New USB device found, idVendor=2109, idProduct=2817, bcdDevice=37.23
[... 45 lines cut ...]
[  633.193797] cdc_ncm 2-1.4.1:1.5: setting tx_max = 16384
[  633.194288] cdc_ncm 2-1.4.1:1.5 eth0: register 'cdc_ncm' at usb-0000:05:00.3-1.4.1, CDC NCM (SEND ZLP), 0c:37:96:a5:86:7d
[  633.233127] cdc_ncm 2-1.4.1:1.5: setting rx_max = 65535
[  633.233544] cdc_ncm 2-1.4.1:1.5: setting tx_max = 65535
[  633.233854] cdc_ncm 2-1.4.1:1.5 enx0c3796a5867d: renamed from eth0
[  635.701086] usb 1-1.4.3: new full-speed USB device number 25 using xhci_hcd
[  635.876034] usb 1-1.4.3: New USB device found, idVendor=28e9, idProduct=0380, bcdDevice= 1.00
[  635.876049] usb 1-1.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  635.876055] usb 1-1.4.3: Product: GD32-USB_Keyboard
[... cut ...]
[  645.952888] usb 1-1.4.3: SerialNumber: VMM_6.05.000
[  646.093627] hid-generic 0003:413C:B06E.000B: hiddev1,hidraw2: USB HID v1.11 Device [BizLink Fischer Cos-2 Controller] on usb-0000:05:00.3-1.4.3/input0
[  646.101490] hid-generic 0003:413C:B06E.000C: hiddev2,hidraw3: USB HID v1.11 Device [BizLink Fischer Cos-2 Controller] on usb-0000:05:00.3-1.4.3/input1
[  646.150856] hid-generic 0003:413C:B06E.000D: hiddev1,hidraw2: USB HID v1.11 Device [BizLink Fischer Cos-2 Controller] on usb-0000:05:00.3-1.4.3/input0
```

Barabazs, 2024-05-02:
It's a known displaylink bug, but unrelated to this project. You'll have to take it up with them.

closed 2024-05-02 by Barabazs