AdnanHodzic/displaylink-debian #913: Kali Linux and PopOS bot do not detect displays -- EVDI not found

whateverwilson, 2024-04-08:
Distro: Kali
Release: kali-rolling
Kernel: 6.6.9-amd64

---------------- DisplayLink info ----------------

Driver version: 1.14.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Raptor
AD107M
VGA: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
NVIDIA Corporation AD107M [GeForce RTX 4060 Max-Q / Mobile] (rev a1)
VGA (3D): 
X11 version: 21.1.11-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xrdp/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 5 outputs: 6 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

4SH-gaupee, 2024-04-09:
Hi, could you please retry with the last release that just been made? https://github.com/AdnanHodzic/displaylink-debian/releases/tag/v5.8.0
This should fix the EVDI problem

Barabazs, 2024-05-02:
The installation clearly failed. Please try reinstalling and if the issue persists, post the output of the installation procedure.