AdnanHodzic/displaylink-debian #916: sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory

jonathanelscpt, 2024-04-09:
Steps take:

1. Uninstalled prior version (i believe 5.3.1)
1. Installed new version (5.8)
1. Installation failed

Installation fails with:

> Performing post install steps
> 
> sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory 

What can I do to fix or troubleshoot this missing file?

```
Select a key: [i/d/r/u/q]: I

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.19.0-41-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev:amd64 is installed
libelf-dev:amd64 is installed
git is installed
pciutils is installed
build-essential is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/5591?filetype=exe
Do you accept?: [Y/n]: Y

Downloading DisplayLink Ubuntu driver:

--2024-04-09 16:20:51--  https://www.synaptics.com/sites/default/files/exe_files/2023-08/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.8-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 54.245.106.105
Connecting to www.synaptics.com (www.synaptics.com)|54.245.106.105|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 18341463 (17M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.8.zip’

DisplayLink_Ubuntu_5.8.zip                           100%[=====================================================================================================================>]  17.49M  3.15MB/s    in 7.0s    

2024-04-09 16:20:59 (2.50 MB/s) - ‘DisplayLink_Ubuntu_5.8.zip’ saved [18341463/18341463]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.8.zip
  inflating: 5.8/DisplayLink USB Graphics Software for Ubuntu 5.8.0-63.33-Release Notes.txt  
  inflating: 5.8/displaylink-driver-5.8.0-63.33.run  
Creating directory displaylink-driver-5.8.0-63.33
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.8.0-63.33  100%  
patching file 5.8/displaylink-driver-5.8/displaylink-installer.sh
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 239 (offset 2 lines).

Installing driver version: 5.8

Distribution discovered: Ubuntu 22.04.4 LTS
WARNING: Version 1.13.1 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory
```

jonathanelscpt, 2024-04-09:
Can't uninstall either:

Select a key: [i/d/r/u/q]: R

```
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.19.0-41-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev:amd64 is installed
libelf-dev:amd64 is installed
git is installed
pciutils is installed
build-essential is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

bash: /opt/displaylink/displaylink-installer.sh: No such file or directory

-------------------------------------------------------------------

```

jonathanelscpt, 2024-04-09:
Update:

Used this to solve the uninstall issue, editing for 5.8:
https://github.com/AdnanHodzic/displaylink-debian/issues/380

Will re-test the installation

Barabazs, 2024-05-03:
afaik this happens when you don't reboot between uninstalling and installing.