AdnanHodzic/displaylink-debian #918: not in the list

ey3sXx, 2024-04-18:


ey3sXx, 2024-04-18:
┌─[✗]─[simone@parrot]─[~/displaylink-debian]
└──╼ $sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/post-install-guide.md [y/N] y

Did you read Troubleshooting most common issues? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/common-issues.md [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: lory
Kernel: 6.5.0-13parrot1-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: 
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Barcelo (rev c1)
VGA (3D): 
X11 version: 21.1.7-3+deb12u7
X11 configs: /etc/X11/xorg.conf.d/00-keyboard.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: File o directory non esistente
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x54 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 2 associated providers: 0 name:AMD Radeon Graphics @ pci:0000:04:00.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Barabazs, 2024-05-03:
This has been resolve with the latest release. Please retry.

closed 2024-05-03 by Barabazs