AdnanHodzic/displaylink-debian #923: ASUS MB16AHV issues connecting on MSI laptop running Ubuntu 22.04

BnZel, 2024-04-26:
Hello,

I have an **ASUS MB16AHV** external monitor that I am connecting to my laptop through the **USB-C port**. My laptop specs are as follows using the neofetch command:

```
OS: Ubuntu 22.04.4 LTS x86_64
Host: Katana GF66 11UE REV:1.0
Kernel: 6.5.0-28-generic
CPU: 11th Gen Intel i7-11800H (16) @ 4.600GHz
GPU: NVIDIA GeForce RTX 3060 Mobile / Max-Q
GPU: Intel TigerLake-H GT1 [UHD Graphics]
```
The ``` xrandr --listproviders ``` did not list another provider and so I removed the ``` /etc/X11/xorg.conf.d/20-displaylink.conf ``` file and followed the instructions on the **Display Detection** section on the **Post Installation Guide**.  After the reboot I logged in with **Xorg**. However this did not work as well.

I retried the installation but get the same results, below is my current situation

The debug info:
```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/post-install-guide.md [y/N] y

Did you read Troubleshooting most common issues? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/common-issues.md [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: jammy
Kernel: 6.5.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 1.14.2
DisplayLink service status: up and running
EVDI service version: 1.14.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
Ltd.
VGA: Intel Corporation TigerLake-H GT1 [UHD Graphics] (rev 01)
NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 21.1.4-2ubuntu1.7~22.04.10
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 5 outputs: 3 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Any help is much appreciated.
It would be a shame to put this back on the shelf!

BnZel, 2024-04-29:
For my HDMI issue, I have managed to get the monitor working through this solution: https://askubuntu.com/questions/1451002/ubuntu-22-04-external-monitor-not-detected

However the issue with just the USB C still remains.

Yoann-86, 2024-05-18:
Hi ! 
I have a similar problem with my HP Elitebook, I'm using Ubuntu 24.04 and I have an external screen, the **Asus MB16AC** connected through the **usb-c** port like you. 
I tried using **displayLink** drivers and when I run `lsusb`  it is recognized and powered.
However, the screen is always black. I don't have a GPU on my laptop. 

`Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 046d:c539 Logitech, Inc. Lightspeed Receiver
Bus 001 Device 003: ID 046d:0aba Logitech, Inc. PRO X Wireless Gaming Headset
Bus 001 Device 004: ID 0bda:5412 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 005: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 006: ID 17e9:4374 **DisplayLink MB16AC**
Bus 001 Device 007: ID 05c8:03b1 Cheng Uei Precision Industry Co., Ltd (Foxlink) Webcam
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
`

When I try with another screen via HDMI, it works well.
Did you find a solution for this issue ?

BnZel, 2024-05-18:
> Hi ! I have a similar problem with my HP Elitebook, I'm using Ubuntu 24.04 and I have an external screen, the **Asus MB16AC** connected through the **usb-c** port like you. I tried using [...]

Hi. Unfortunately I couldn't find a workaround, I've asked around and it seems like DisplayLink is a hassle to work with. Especially if one has a Nvidia driver.

Are you sure your device does not have a GPU? If not, then I don't think it would work at all. What is your specs and debug info?

Yoann-86, 2024-05-20:
> Hi. Unfortunately I couldn't find a workaround, I've asked around and it seems like DisplayLink is a hassle to work with. Especially if one has a Nvidia driver.
Are you sure your device does not have a GPU? If not, then I don't think it would work at all. What is your specs and debug info?

Hi ! I solved my issue by reinstalling displaylink and using another cable for the connection. This only works with a usb-a (ss) to usb-c. I found this solution on the displaylink forum [in this page](https://www.displaylink.org/forum/showthread.php?t=66809&highlight=MB16AC).

Thank you for helping and I hope this solution will solve your issue.