AdnanHodzic/displaylink-debian #925: Displaylink not working on Ubuntu 24.04 (getting error on xrandr --setprovideroutputsource)

rodmba, 2024-04-28:
Hey guys, could someone help me with this setup?

I'm getting the following error while running `xrandr --setprovideroutputsource 1 0`: 

```
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x43
  Serial number of failed request:  20
  Current serial number in output stream:  21
```

My debug stuff:

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: noble
Kernel: 6.8.0-31-generic

---------------- DisplayLink info ----------------

Driver version: 1.14.2
DisplayLink service status: up and running
EVDI service version: 1.14.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: CometLake-U
VGA: Intel Corporation CometLake-U GT2 [UHD Graphics] (rev 02)
VGA (3D): NVIDIA Corporation GP108M [GeForce MX250] (rev a1)
X11 version: 21.1.12-1ubuntu1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x43 cap: 0x0 crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x102 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xe1 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xc0 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x9f cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0x7e cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

-------------------------------------------------------------------
```

darren051, 2024-05-02:
Breaks on every upgrade

kegeg, 2024-06-09:
Did you manage to make it work? Mine also broke when upgrading to 24.04.

MartinPaulo, 2024-07-21:
My upgrade to 24.04 failed spectacularly. I have no idea if the Displaykink driver was the issue. But I was able to get everything to work by walking back to an old kernel version. 
I'm now on 24.04 with a 6.3 kernel, :(. The Displaylink driver seems to be working Ok with my setup, but I've wasted a solid day trying to get everything back into a useable state. I'll slowly start walking my kernel versions forward.

wdschei, 2024-10-26:
I finally did the 24.04 Kubuntu update and had the same issue.
``` sh
$ lsb_release -a && uname -r
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.1 LTS
Release:        24.04
Codename:       noble
6.8.0-47-generic
```
I had previously installed the Synaptics repository as per [these instructions]https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu):
```sh
sudo apt install ./Downloads/synaptics-repository-keyring.deb
```
Although it shouldn't have been necessary, the following ultimately fixed it for me:
```sh
sudo apt purge evdi
sudo apt install displaylink-driver
sudo reboot now
```

juliancervos, 2024-11-19:
> Although it shouldn't have been necessary, the following ultimately fixed it for me:
> 
> ```shell
> sudo apt purge evdi
> sudo apt install displaylink-driver
> sudo reboot now
> ```

Tried to follow that, but I get an error when trying to do `sudo apt install displaylink-driver`:
```bash
$ sudo apt install displaylink-driver
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package displaylink-driver

```

wdschei, 2024-11-19:
@juliancervos did you install the Synaptics repository as per [these instructions](https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu):

```shell
sudo apt install ./Downloads/synaptics-repository-keyring.deb
```

juliancervos, 2024-11-19:
Yes, I did, so not sure what went wrong 😕 

I ended up installing it through the official driver directly through the downloadable script instead of through the APT repository.