AdnanHodzic/displaylink-debian #931: Meta issue for 'failed to create symbolic link '/opt/displaylink/....' (Debian)

Barabazs, 2024-05-03:
#805
#855

Potential solution https://github.com/AdnanHodzic/displaylink-debian/issues/805#issuecomment-1634037909

rogue73, 2024-06-06:
Hi Barabazs, I've done quickly a fresh install of debian 12. Then I've comment out the above block and could perfectly install and setup displaylink. 
Also on my existing installation (Debian sid/unstable) I could install displaylink and it doesn't matter if the lines above are comment out or not.

I think these line can be removed.

Cheers
rogue73

Barabazs, 2024-06-07:
Thanks @rogue73!
Can you check if this file exists please? `/usr/lib/x86_64-linux-gnu/libstdc++.so.6`

rogue73, 2024-06-07:
Hi, `/usr/lib/x86_64-linux-gnu/libstdc++.so.6` exist as symlink to `/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30`

AngelBlue01, 2024-09-23:
Hi I ran `mkdir -p /opt/displaylink` and `ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 /opt/displaylink/libstdc++.so.6` as suggested in #805  now I get the error

> sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory

tymek1317, 2024-10-29:
> Hi I ran `mkdir -p /opt/displaylink` and `ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 /opt/displaylink/libstdc++.so.6` as suggested in #805 now I get the error
> 
> > sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory

Same here

nigelsim, 2024-11-07:
In my system (Debian 12 Kernel 6.1.0-26-amd64) my when I get the above issues (`ln` and `displaylink-driver.service`) it because the EVDI build fails, but the installer continues as if it succeeds. 

[install.log](https://github.com/user-attachments/files/17670628/install.log)

Unfortunately, the EVDI build cleans itself up, meaning we lose the logs when run using the installer.

Running the EVDI build by hand, I see the following errors:

```
/var/lib/dkms/evdi/1.14.7/build/evdi_fb.c: In function ‘evdifb_create’:
/var/lib/dkms/evdi/1.14.7/build/evdi_fb.c:408:23: error: ‘struct drm_fb_helper’ has no member named ‘info’
  408 |         efbdev->helper.info = info;
      |                       ^
/var/lib/dkms/evdi/1.14.7/build/evdi_fb.c: In function ‘evdi_fbdev_destroy’:
/var/lib/dkms/evdi/1.14.7/build/evdi_fb.c:468:27: error: ‘struct drm_fb_helper’ has no member named ‘info’
  468 |         if (efbdev->helper.info) {
      |                           ^
/var/lib/dkms/evdi/1.14.7/build/evdi_fb.c:469:38: error: ‘struct drm_fb_helper’ has no member named ‘info’
  469 |                 info = efbdev->helper.info;
      |                                      ^
/var/lib/dkms/evdi/1.14.7/build/evdi_fb.c: In function ‘evdi_fbdev_unplug’:
/var/lib/dkms/evdi/1.14.7/build/evdi_fb.c:560:27: error: ‘struct drm_fb_helper’ has no member named ‘info’
  560 |         if (efbdev->helper.info) {
      |                           ^
/var/lib/dkms/evdi/1.14.7/build/evdi_fb.c:563:38: error: ‘struct drm_fb_helper’ has no member named ‘info’
  563 |                 info = efbdev->helper.info;
      |                                      ^
make[2]: *** [/usr/src/linux-headers-6.1.0-26-common/scripts/Makefile.build:255: /var/lib/dkms/evdi/1.14.7/build/evdi_fb.o] Error 1
```

I hope this isn't a red herring.

nigelsim, 2024-11-08:
If I revert this commit in EVDI then it compiles as expected https://github.com/DisplayLink/evdi/commit/88ef73df576aacdfff6597616a77b88f42eca4c7

To implement, I ran the installer. Once it failed, I extracted `6.1/displaylink-driver-6.1/evdi.tar.gz` patched the `evdi_fb.c` and recompressed, replacing the tar.gz file. Then, ran `./displaylink-installer.sh install` from within `6.1/displaylink-driver-6.1`

I'll file an upstream bug report later.

Just as a note, `uname -a` reports `Linux magrathea 6.1.0-26-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.112-1 (2024-09-30) x86_64 GNU/Linux`. On first glance I read this as running Kernel version `6.1.0` but it is actually `6.1.112`, which is why I needed revert the commit because it incorrectly switched to the new DRM struct at version 6.1.51 when it isn't actually introduced until 6.2.0 https://github.com/torvalds/linux/commit/9877d8f6bc374912b08dfe862cddbb78b395a5ef

matteoopenf, 2024-11-11:
any news I have the same errore on debian 12  PREEMPT_DYNAMIC Debian 6.1.115-1 (2024-11-01) x86_64 GNU/Linux ?

matteoopenf, 2024-11-12:
> /usr/lib/x86_64-linux-gnu/libstdc++.so.6

in my case exist

matteoopenf, 2024-11-12:
in my case try to find the file here ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory
not in the directory you mean above

mulderij, 2024-11-13:
> in my case try to find the file here ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory not in the directory you mean above

As mentioned in https://github.com/AdnanHodzic/displaylink-debian/issues/931#issuecomment-2444495660 this is because the dir /opt/displaylink doesn't exist. However creating that dir results in the next error about `sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory`

matteoopenf, 2024-11-13:
> As mentioned in https://github.com/AdnanHodzic/displaylink-debian/issues/931#issuecomment-2444495660 this is because the dir /opt/displaylink doesn't exist. However creating that dir results in the [...]

Mmm and there is a fix or a way to creare the missing file and directory?

nigelsim, 2024-11-30:
Upstream issues raised https://github.com/DisplayLink/evdi/issues/499

eBug, 2024-12-16:
#961 throws the same `ln` error

eBug, 2024-12-16:
> #805 #855
> 
> **Description:** Installation on Debian often fails with `ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory`
> 
> **Potential hacky workaround:** [#805 (comment)](https://github.com/AdnanHodzic/displaylink-debian/issues/805#issuecomment-1634037909)
> 
> **Root cause:** this code block from 2017 which may or may not be needed anymore...
> 
> https://github.com/AdnanHodzic/displaylink-debian/blob/512d29bd9e76cf694dcabd19c83e6a3a63242ffd/displaylink-debian.sh#L425-L430
> 
> I don't have access to a Debian install and don't have the capacity to troubleshoot this on a VM. I would appreciate it if someone can try do comment out the code block and perform the installation on a fresh Debian install. (so not tainted with a previous displaylink install)

I just tried your workaround on my Debian 12 install.

As others pointed out, there's no more `ln` error but it's sed complaining now:

`sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory`

eBug, 2024-12-16:
> If I revert this commit in EVDI then it compiles as expected [DisplayLink/evdi@88ef73d](https://github.com/DisplayLink/evdi/commit/88ef73df576aacdfff6597616a77b88f42eca4c7)
> 
> To implement, I ran the installer. Once it failed, I extracted `6.1/displaylink-driver-6.1/evdi.tar.gz` patched the `evdi_fb.c` and recompressed, replacing the tar.gz file. Then, ran `./displaylink-installer.sh install` from within `6.1/displaylink-driver-6.1`
> 
> I'll file an upstream bug report later.
> 
> Just as a note, `uname -a` reports `Linux magrathea 6.1.0-26-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.112-1 (2024-09-30) x86_64 GNU/Linux`. On first glance I read this as running Kernel version `6.1.0` but it is actually `6.1.112`, which is why I needed revert the commit because it incorrectly switched to the new DRM struct at version 6.1.51 when it isn't actually introduced until 6.2.0 [torvalds/linux@9877d8f](https://github.com/torvalds/linux/commit/9877d8f6bc374912b08dfe862cddbb78b395a5ef)

reverting the commit you mention by hand, and re-running `./displaylink-installer.sh install` got the module compiled here. Thx.

nigelsim, 2024-12-22:
The upstream issue has been solved in evdi 1.14.8. Because the embedded `evdi.tar.gz` has the top level directory stripped out, the simplest may be to apply the following patch to this repo, which I adapted from #955

```
diff --git a/displaylink-debian.sh b/displaylink-debian.sh
index 68f5947..3e83a2a 100755
--- a/displaylink-debian.sh
+++ b/displaylink-debian.sh
@@ -368,6 +368,15 @@ then
        ln -sf /lib/modules/$(uname -r)/build/Makefile /lib/modules/$(uname -r)/build/Kconfig
 fi
 
+# Replace evdi with newer version
+evdi_ver=1.14.8
+echo 'Replacing evdi...'
+mkdir evdi
+wget https://github.com/DisplayLink/evdi/archive/refs/tags/v${evdi_ver}.tar.gz
+tar -xzf v${evdi_ver}.tar.gz --strip-components=1 -C evdi
+tar -czf $driver_dir/displaylink-driver-${version}/evdi.tar.gz -C evdi .
+rm -rf evdi v${evdi_ver}.tar.gz
+
 # Patch displaylink-installer.sh to prevent reboot before our script is done.
 patchName="displaylink-installer.patch"
 finalPatchPath="$resourcesDir$patchName"
```

Probably unrelated, but on my Intel NUC, using the `displaylink-debian.sh` I had to update my `/etc/X11/xorg.conf.d/20-displaylink.conf` to which I didn't have to do previously when I ran the patched `./displaylink-installer.sh install`:

```
Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

mulderij, 2024-12-23:
> Probably unrelated, but on my Intel NUC, using the `displaylink-debian.sh` I had to update my `/etc/X11/xorg.conf.d/20-displaylink.conf` to which I didn't have to do previously when I ran the [...]

Yes, this is unrelated. However it would be nice if this could be fixed for Intel Video. Currently I always edit this by hand after running the setup.
[https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348)

TJNII, 2025-02-09:
Great work, @nigelsim.  Once I followed your workaround and rebooted the displaylink monitors "just worked".  I was also happy to see that some manual xrandr steps I had documented on a different machine from mid 2023 were no longer needed, I was very pleasantly surprised with how good the user experience on Debian 12 was once the evdi issue was overcome.  Thanks for figuring this out and reporting it upstream.

omartinezcelis, 2025-03-22:
Sorry to ask, I really don't understand the steps to follow to do the workaround, I copy the code suggested in 
#955 but still not good results

Lingatsu, 2026-08-27:
> Sorry to ask, I really don't understand the steps to follow to do the workaround, I copy the code suggested in [#955](https://github.com/AdnanHodzic/displaylink-debian/issues/955) but still not good results

Hi, in the case it can help you or someone else despite the one year gap, I'll share the steps I followed to make the displaylink-debian drivers works on my config : 
```
#Distro: Debian
Release: trixie
Kernel: 6.12.105+deb13-amd64
```
I had DKMS or libstdc++ missing error message and also the` --uninstall` script was not removing the previous contents. The main problem was that I didn't had the "/usr/sbin" path directory configured when launching the displaylink-debian.sh.*

**_I did all theses steps in su privileges on he displaylink-debian root folder._** 

Steps : 
1. edit `/root/.bashrc file` and add this line at the end `export PATH=/usr/sbin:$PATH`
2. close/re-open the terminal and write `echo $PATH` to verify if `/usr/sbin` appear now
3. `./displaylink-debian.sh --uninstall ` and reboot
4.  `./displaylink-debian.sh --install ` and reboot
5. `./displaylink-debian.sh --debug ` and see verify if the driver's running :
```
---------------- DisplayLink info ----------------

Driver version: 1.15.0
DisplayLink service status: up and running
EVDI service version: 1.15.0
```
6. if like mine, your external monitor still didn't work at this step : replace the `/etc/X11/xorg.conf.d/20-displaylink.conf` file's content with that one** :            
```Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```
7. reboot and enjoy      

Sources :
*. As mentionned by @AysadKozanoglu in [#61](https://github.com/AdnanHodzic/displaylink-debian/issues/61#issuecomment-664954861)_
**. As mentionned by @AdnanHodzic in [#228](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348)_