AdnanHodzic/displaylink-debian #932: It is not working on Ubuntu 23.10 - Can someone help me?

FelipeGerolomo, 2024-05-12:
--------------- Linux system info ----------------

Distro: Ubuntu
Release: mantic
Kernel: 6.5.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 1.14.2
1.14.4
DisplayLink service status: up and running
EVDI service version: 1.14.4

------------------ Graphics card -----------------

Vendor: i915
Subsystem: TigerLake-LP
VGA: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 03)
VGA (3D): 
X11 version: 21.1.7-3ubuntu2.9
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

heredia21, 2024-05-22:
The fix for me was switching to Wayland.

rogue73, 2024-06-08:
Hi, I think, you need to put the following into /etc/X11/xorg.conf.d/20-displaylink.conf

Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

after installing displaylink. It seems that since kernel 6.6 and above the detection of graphic card subsystem fails and the script add the wrong content to xorg config file.

rogue73

5472qaywsx, 2024-06-19:
> Hi, I think, you need to put the following into /etc/X11/xorg.conf.d/20-displaylink.conf
>
> Section "OutputClass" Identifier "DisplayLink" MatchDriver "evdi" Driver "modesetting" Option "AccelMethod" [...]

Thank you, that worked for me on debian 12