AdnanHodzic/displaylink-debian #937: Unable to install DisplayLink on Debian (because of evdi)

zineb-ch, 2024-06-05:
Hi,

I am unable to install DisplayLink on my Debian machine. I could not install evdi...

Here is what I have :

-------------- Linux system info ----------------

Distro: Debian
Release: bookworm
Kernel: 6.1.0-21-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0
1.12.0+dfsg
1.14.4
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

--------------------------------------------------------

Could someone help me please ?

jbize, 2024-11-05:
I'm having same issue on Debian 12 (bookworm).  Did anyone get it resolved?

coldsubstance-deb, 2025-04-03:
https://github.com/AdnanHodzic/displaylink-debian/issues/656#issuecomment-996808332
this is the current solution i think?  its really janky but it probably works