AdnanHodzic/displaylink-debian #941: Monitors not detected (StarTech USB32HD2, Ubuntu 24.04)

greengreengrassofhome, 2024-06-10:
I'm trying to connect two monitors to a laptop with an USB 3 port via [a StarTech USB32HD2 adapter](https://www.startech.com/en-gb/display-video-adapters/usb32hd2). The official product description claims the device only supports Windows, however some quick searching indicates others have gotten it working under Ubuntu before (see for example https://askubuntu.com/questions/1169583/how-to-use-startech-usb32hd2-with-ubuntu-18-04).

I have installed `displaylink-debian`, and the debug output tells me that it is up and running. However the monitors remain dark, and when I for example open Displays in the system settings, it does not show me any monitors except the built-in one.

This is the output after I ran `sudo ./displaylink-debian.sh --debug` for the first time:
```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: noble
Kernel: 6.8.0-35-generic

---------------- DisplayLink info ----------------

Driver version: 1.14.4
DisplayLink service status: up and running
EVDI service version: 1.14.4

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: [Radeon
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso/Raven 2 [Radeon Vega Series / Radeon Vega Mobile Series] (rev ce)
VGA (3D): 
X11 version: 21.1.12-1ubuntu1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x53 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:AMD Radeon Vega 3 Graphics @ pci:0000:03:00.0
```

And this is the output after I removed the `/etc/X11/xorg.conf.d/20-displaylink.conf` file according to the post installation guide:

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: noble
Kernel: 6.8.0-35-generic

---------------- DisplayLink info ----------------

Driver version: 1.14.4
DisplayLink service status: up and running
EVDI service version: 1.14.4

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: [Radeon
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso/Raven 2 [Radeon Vega Series / Radeon Vega Mobile Series] (rev ce)
VGA (3D): 
X11 version: 21.1.12-1ubuntu1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x53 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:AMD Radeon Vega 3 Graphics @ pci:0000:03:00.0
Provider 1: id: 0xe0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xbf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x9e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x7d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

I've also attached the output from [dmesg.txt](https://github.com/user-attachments/files/15765191/dmesg.txt) and the relevant info for the adapter from [lsusb.txt](https://github.com/user-attachments/files/15765192/lsusb.txt).

I've read the post installation and troubleshooting guides, and don't know where to look next. Any pointers would be much appreciated.

greengreengrassofhome, 2024-06-10:
Looking at `/var/log/syslog` , I see a few instances of the following:

```
/usr/libexec/gdm-x-session[2166]: MESA-LOADER: failed to open evdi: /usr/lib/dri/evdi_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
```

May that be relevant?

Additionally, I installed Ubuntu 22.04 alongside 24.04 and did the whole procedure again, with the exact same result (including the error messages mentioned above).

Barabazs, 2025-03-23:
afaik your device does not have a displaylink chip,so won't work with displaylink.

closed 2025-03-23 by Barabazs