AdnanHodzic/displaylink-debian #944: Debian 12.5, system crashes at reboot

jlgarnier, 2024-07-01:
Hi all,

Tried to install the software on a fresh Debian 12.5 Dell Latitude 5480 machine: everything installed properly but the system crashes (even in rescue mode) after the post-install reboot : fifo: SCHED_ERROR 06 [], which ends in "DRM: failed to idle channel 1 [DRM]".

I have obviously no chance to get any additional information from the system as all I can do is switch the machine off...

Did anyone ever get the same issue?

Thanks in advance for any help!

thool2b3, 2024-08-28:
Hi jlgarnier,
I experienced the same and on my first attempt I activated an older Timeshift snapshot to get my system up and running again.
When it happened the second time I followed the advice in https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/post-install-guide.md and renamed `/etc/X11/xorg.conf.d/20-displaylink.conf` and my system started again without rewinding the snapshot. But my external monitor is still not working see the issue here:
https://github.com/AdnanHodzic/displaylink-debian/issues/952

jlgarnier, 2024-09-09:
Thanks for the feedback! I finally found that the issue came from a secondary GPU (Nvidia), which was partially supported by Debian and collided with the default embedded GPU (Intel). Therefore, I deactivated the 2nd GPU and got the expected boot sequence up to a working desktop!

However, now that everything was working pretty, I tried to install the software again. Installation went perfectly fine but ended in a boot issue again: the machine now indicates "unable to start lightdm.service" then stops. I've opened another issue on this forum to address it.

Any help would be deeply appreciated!

closed 2024-09-13 by jlgarnier