AdnanHodzic/displaylink-debian #955: evdi 1.14.4 won't compile on kernel 6.10

cilynx, 2024-09-22:
It's a [known issue](https://github.com/DisplayLink/evdi/issues/469) that has been fixed in 1.14.5, but 1.14.4 ships with the Ubuntu DisplayLink driver.  To work around the issue, I added a few lines right before the patch is applied in https://github.com/AdnanHodzic/displaylink-debian/blob/b763e8fec501243b6750094f4fb3fc452668b7f9/displaylink-debian.sh#L370 to replace `evdi.tar.gz` that ships with the driver with a stripped version of https://github.com/DisplayLink/evdi/archive/refs/tags/v1.14.6.tar.gz.

```
# Replace evdi with newer version
echo 'Replacing evdi...'
mkdir evdi
tar -xzf evdi-1.14.6.tar.gz --strip-components=1 -C evdi
tar -czf $driver_dir/displaylink-driver-${version}/evdi.tar.gz -C evdi .
rm -rf evdi
```

Not sending a PR as this is a lame hack, but I can confirm it works on Trixie's 6.10.9

cilynx, 2024-09-22:
Slightly more robust:
```
# Replace evdi with newer version
evdi_ver=1.14.6
echo 'Replacing evdi...'
mkdir evdi
wget https://github.com/DisplayLink/evdi/archive/refs/tags/v${evdi_ver}.tar.gz
tar -xzf v${evdi_ver}.tar.gz --strip-components=1 -C evdi
tar -czf $driver_dir/displaylink-driver-${version}/evdi.tar.gz -C evdi .
rm -rf evdi v${evdi_ver}.tar.gz

gsoul, 2025-03-12:
Just in case somebody would be wondering about how to apply solution above. Here's where I inserted it into the displaylink-debian.sh to fix the issue (wasn't obvious for me from the start).

```
# ...existing code...
mv displaylink-driver-${version}*/ $driver_dir/displaylink-driver-${version}
# get sysinitdaemon
sysinitdaemon=$(sysinitdaemon_get)

# Replace evdi with newer version
evdi_ver=1.14.8
echo 'Replacing evdi with version' $evdi_ver '...'
mkdir -p evdi
wget https://github.com/DisplayLink/evdi/archive/refs/tags/v${evdi_ver}.tar.gz
tar -xzf v${evdi_ver}.tar.gz --strip-components=1 -C evdi
tar -czf $driver_dir/displaylink-driver-${version}/evdi.tar.gz -C evdi .
rm -rf evdi v${evdi_ver}.tar.gz

# modify displaylink-installer.sh
sed -i "s/SYSTEMINITDAEMON=unknown/SYSTEMINITDAEMON=$sysinitdaemon/g" $driver_dir/displaylink-driver-${version}/displaylink-installer.sh
# ...existing code...
```