AdnanHodzic/displaylink-debian #965: kali 6.10.11 -- Fixed

Xero12369, 2024-10-21:
I was having the same issues other people were having. Here is the process I used and my display link is working again on kali 6.10.11

sudo git clone https://github.com/DisplayLink/evdi.git
cd evdi
cd module
sudo ./dkms_install.sh
sudo mkdir -p /opt/displaylink

Head over to the display link website and download the Ubuntu driver

https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu
sudo dpkg -i synaptics-repository-keyring.deb
sudo apt-get install -f

Then I rerran the install script from displaylink-debian
sudo ./displaylink-debian.sh

successful install and upon reboot, everything was working.

renoproc, 2024-10-22:
Worked for me too :+1:

beelow2010, 2024-11-04:
Heyo, unfortunately I still have some weird issues (using a HP Universal Dock G2).

First I tried to uninstall everything related to displaylink and evdi (incl. reboot) before attempting your solution.

However, when performing the installation of DisplayLink-Ubuntu I get an issue in the post install step:

```
Installing driver version: 6.1

Distribution discovered: Kali GNU/Linux Rolling
WARNING: This is not an officially supported distribution.
Please use DisplayLink Forum for getting help if you find issues.
WARNING: Version 1.14.7 of EVDI kernel module is already running.
Please uninstall all other versions of DisplayLink Linux Software before attempting to install.
Installation terminated.

-------------------------------------------------------------------

Performing post install steps

sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory
```

When looking at the providers, everything seems to be as expected but /etc/X11/xorg.conf.d/20-displaylink.conf was not created...

```
Starting Debug ...

Did you read Post Installation Guide? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/post-install-guide.md [y/N] y

Did you read Troubleshooting most common issues? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/common-issues.md [y/N] y

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 6.10.11-amd64

---------------- DisplayLink info ----------------

Driver version: 1.14.7
DisplayLink service status: 
EVDI service version: 1.14.7

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: 
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Phoenix1 (rev dd)
VGA (3D): 
X11 version: 21.1.13-3~kali1                                                                                                                                                                                                                 
                                                                                                                                                                                                                                             
-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x4a cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 8 associated providers: 4 name:modesetting
Provider 1: id: 0x182 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x161 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x140 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x11f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Is someone experiencing a similar issue?

Thanks in advance!