AdnanHodzic/displaylink-debian #975: Make error Bad return status for module build on kernel

andrewtoups, 2024-12-02:
I am getting the same error listed on the common errors page.

However, running the suggested solution does not resolve the issue. Running the install script results in the same error.

Debug info:

```
--------------- Linux system info ----------------

Distro: Debian
Release: bookworm
Kernel: 6.1.0-28-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0+dfsg
1.14.2
1.14.7
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
VGA: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 03)
VGA (3D): 
X11 version: 21.1.7-3+deb12u8

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

andrewtoups, 2024-12-02:
I should note that I ran `--debug` while running a wayland session, but it's my understanding that this should work in wayland as well, even if the debug steps rely on x11?

In either case, this is the output when I log in to an x11 session:

```
--------------- Linux system info ----------------

Distro: Debian
Release: bookworm
Kernel: 6.1.0-28-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0+dfsg
1.14.2
1.14.7
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
VGA: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 03)
VGA (3D): 
X11 version: 21.1.7-3+deb12u8

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

andrewtoups, 2024-12-02:
Also -- not sure how relevant this is, but when I try to install, the message says:
```
Consult /var/lib/dkms/evdi/1.14.7/build/make.log for more information.
```

However, on my system there is no `/var/lib/dkms/evdi/1.14.7`. Instead, I have `/var/lib/dkms/evdi/1.14.2`. When I try to update the evdi package using apt, it says it's at the latest version. If I specify the version with this command:

`apt-get install evdi-dkms=1.14.7`

It says:

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package evdi-dkms is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Version '1.14.7' for 'evdi-dkms' was not found
```

I could be barking up the wrong tree with this, but it seems like it could be related to the issue here?

andrewtoups, 2024-12-02:
Update: I was able to work around this by hard-coding the `$version` variable to `"6.0"`, which as of this writing is the next-to-latest release. (The current latest release is `6.1`).

Here's the quick and dirty change to `displaylink-debian.sh` if anyone else has this issue and just wants to get things working: 
```diff
 # Get latest versions
 versions=$(wget -q -O - https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu | grep "<p>Release: " | head -n 2 | perl -pe '($_)=/([0-9]+([.][0-9]+)+(\ Beta)*)/; exit if $. > 1;')
 # if versions contains "Beta", try to download previous version
 if [[ $versions =~ Beta ]]; then
-     version=$(wget -q -O - https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu | grep "<p>Release: " | head -n 2 | perl -pe '($_)=/([0-9]+([.][0-9]+)+(?!\ Beta))/; exit if $. > 1;')
+     # version=$(wget -q -O - https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu | grep "<p>Release: " | head -n 2 | perl -pe '($_)=/([0-9]+([.][0-9]+)+(?!\ Beta))/; exit if $. > 1;')
+    version="6.0"
     dlurl="https://www.synaptics.com/$(wget -q -O - https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu | grep -B 2 $version'-Release' | perl -pe '($_)=/<a href="\/([^"]+)"[^>]+class="download-link"/')"
     driver_url="https://www.synaptics.com/$(wget -q -O - ${dlurl} | grep '<a class="no-link"' | head -n 1 | perl -pe '($_)=/href="\/([^"]+)"/')"
 else
-    version=`wget -q -O - https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu | grep "<p>Release: " | head -n 1 | perl -pe '($_)=/([0-9]+([.][0-9]+)+)/; exit if $. > 1;'`
+    # version=`wget -q -O - https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu | grep "<p>Release: " | head -n 1 | perl -pe '($_)=/([0-9]+([.][0-9]+)+)/; exit if $. > 1;'`
+    version="6.0"
     dlurl="https://www.synaptics.com/$(wget -q -O - https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu | grep -B 2 $version'-Release' | perl -pe '($_)=/<a href="\/([^"]+)"[^>]+class="download-link"/')"
     driver_url="https://www.synaptics.com/$(wget -q -O - ${dlurl} | grep '<a class="no-link"' | head -n 1 | perl -pe '($_)=/href="\/([^"]+)"/')"
fi
```

If you're going to try this yourself, copy the version number the script reports when it tries to install the display link driver, and then [check the **Legacy Drivers** section of the release page](https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu) to find the next-to-most-recent release, and use that when you modify the shell script.

It's worth noting that for me, with driver version `6.0` installed in X11 mode, plugging in the external dock would not register any new devices, and while directly connected monitors would receive a signal, the entire desktop environment would freeze as soon as this happened. I couldn't even boot into the OS with the screen connected.

However, when I switched to Wayland, everything works again as it used to, except the dock now also works correctly.

I'm leaving this issue open (if that's okay) because I'd still like to understand why using version `6.1` didn't work. I'm guessing that my kernel headers aren't using the version of `evdi-dkms` the driver was looking for? If anyone reading this has a better understanding of the hardware/driver side of Linux, please correct me if I'm wrong.

At the very least, I think it would be worth letting the user manually select the version if they so choose, either via a prompt or a command line option, and then add a note in the **Common issues** page to try a previous version if you encounter the `make` error. It's my understanding that this project isn't actively maintained right now, so I'll see if I can find a way to modify the script to make that work.

mschmitt, 2024-12-03:
This is the first workaround that works for me. And it also explains why a few of my users who installed a few months ago, don't have any issues. They're simply still on 6.0.

mschmitt, 2024-12-04:
Hotfix fork with a somewhat reasonable workaround:

```
git clone https://github.com/mschmitt/displaylink-debian.git
cd displaylink-debian
sudo DISPLAYLINK_VERSION=6.0 ./displaylink-debian.sh
```

Amondale, 2024-12-14:
The workaround worked under LMDE (same kernel), and I finally got it to install evdi ok (see debug info below), BUT ... seems to be sending a signal to the monitor, cycles periodically and monitor responds with "no signal detected" then apparently gets a signal, but never detected under display settings. 

Any ideas? I had an HDMI monitor plugged in while running the debug param of the modified displaylink-debian.sh script. Worth mentioning I have Secure Boot on and enabled, but no run of mokutil happened, as it did reliably with Ubuntu and derivatives (actually got Pop!Os to load the DD driver and even boot under Secure Boot!)

I do wish there was a way to get evdi 1.14.7 to install independently of the displaylink package, but I guess it's "their" package, even though it's open source and distributed by Debian.

```

--------------- Linux system info ----------------

Distro: Linuxmint
Release: faye
Kernel: 6.1.0-28-amd64

---------------- DisplayLink info ----------------

Driver version: 1.14.4
DisplayLink service status: up and running
EVDI service version: 1.14.4

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Vega
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev c2)
VGA (3D): 
X11 version: 21.1.7-3+deb12u8
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x56 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:AMD Radeon Graphics @ pci:0000:04:00.0

-------------------------------------------------------------------

```

Amondale, 2024-12-14:
The 6.0 version of displaylink and Secure Boot evidently don't get along, even if `mokutil --disable-validation` is run disabling shim signing. So it appears we just wait for displaylink to figure out what's going on with the installation routine, and / or see if we can get Adnan to review and perhaps find out what is causing the evdi failures in 1.14.7 which mayt support the 6.1 kernels under Secure Boot.  It seems it's all figured out for Ubuntu 24.04 even with SB enabled.

andrewtoups, 2024-12-16:
Yeah, and I'm encountering visual glitches using driver version 6.0 on my debian system, both on the software and hardware level, that did not occur before installing these drivers.

They work well enough but for anyone reading this just know that this is an imperfect solution. I'm just a web developer so there's a limit what I can understand about what's going wrong here, but again if anyone reading this has more experience with doing that kind of work, it would be great if we could at least identify the underlying issues here. It's dismaying (though completely understandable) that modern linux desktops still have to deal with basic compatibility issues like this.

mschmitt, 2024-12-16:
We tested Displaylink versions 6.0 and 6.1 on Secureboot without any issue; however we may have hooked dkms module signing with our platform key in a way that Synaptics accidentally doesn't choke on.

After a brief affair with my forked repository (above) and Displaylink 6.0, we started to use Adnan's installer with Displaylink 6.1 as-is, with the Debian Backport Kernel 6.11, which is completely beyond the Linux 6.1/6.2 situation with evdi.