AdnanHodzic/displaylink-debian #977: Debian 13 - 6.12.5-amd64

fmrtn, 2024-12-24:
It works fine in Debian 13 with 6.11.10-amd64 kernel, with Xorg or Wayland. But it doesn't work with 6.12.5 with either Xorg or Wayland.

I've tried the various options recommended for /etc/X11/xorg.conf.d/20-displaylink.conf

--------------- Linux system info ----------------

Distro: Debian
Release: trixie
Kernel: 6.12.5-amd64

---------------- DisplayLink info ----------------

Driver version: 1.14.7
DisplayLink service status: up and running
EVDI service version: 1.14.7

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Device
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 21.1.14-2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x48 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 7 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Mathew3000, 2025-01-13:
I had the same issue, it seems to be a problem with the `evdi` module. according to this: https://github.com/DisplayLink/evdi/releases the newer versions should support both 6.12 and 6.13 kernel.

System: Debian Trixie - Framework 16 - Wayland

I got it to work with `evdi 1.14.8`, and 6.12.8-amd64 here is what I did:
- booted 6.11.10 kernel
- installed displaylink-debian 
- reboot to 6.11.10 to check if everything works: it does
- boot 6.12.8 kernel -> stops working as expected
- installed `evdi-dkms` 1.14.8 via `apt` -> reboot
- installed displaylink-debian with `install` instead of `re-install` -> reboot
-> WORKS!

Not sure if every reboot is required but this is all I did

jezh42, 2025-01-14:
@Mathew3000 This fix worked!! I am on 6.12.6 with Debian Trixie. However, I didn't install an old kernel and the final restart wasn't required in my case. What i performed is:

1. Boot into 6.12.6
2. Install displaylink-debian, ensure no errors, reboot.
3. `apt install evdi-dkms`, which currently ships 1.14.8, as per Trixie/Testing, reboot.

Works for both X11 and Wayland, using KDE Plasma 6.2.5