AdnanHodzic/displaylink-debian #982: It didn't work on debian 12

nomevini, 2025-02-06:
--------------- Linux system info ----------------

Distro: Debian
Release: bookworm
Kernel: 6.1.0-30-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0+dfsg
1.14.7
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd
VGA: Intel Corporation HD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 21.1.7-3+deb12u8

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Arquivo ou diretório inexistente
Contents:
 

-------------------- Monitors --------------------

Providers: number : 0

mustekito, 2025-02-12:
Same OS, same problem.

--------------- Linux system info ----------------

Distro: Debian
Release: bookworm
Kernel: 6.1.0-30-amd64

---------------- DisplayLink info ----------------

Driver version: 1.14.7
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Alder
VGA: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
VGA (3D): 
X11 version: 21.1.7-3+deb12u8

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 0

~~~

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_6.1.zip
  inflating: 6.1/displaylink-driver-6.1.0-17.run  
  inflating: 6.1/DisplayLink USB Graphics Software for Ubuntu 6.1.0-17-Release Notes.txt  
Creating directory displaylink-driver-6.1.0
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 6.1.0  100%  
patching file 6.1/displaylink-driver-6.1/displaylink-installer.sh
Hunk #1 succeeded at 215 (offset 3 lines).

Installing driver version: 6.1

Distribution discovered: No LSB modules are available.
Debian GNU/Linux 12 (bookworm)
WARNING: This is not an officially supported distribution.
Please use DisplayLink Forum for getting help if you find issues.

Installing
[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/evdi/1.14.7/source -> /usr/src/evdi-1.14.7

Building module:
Cleaning build area...
make -j16 KERNELRELEASE=6.1.0-30-amd64 all INCLUDEDIR=/lib/modules/6.1.0-30-amd64/build/include KVERSION=6.1.0-30-amd64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-30-amd64 (x86_64)
Consult /var/lib/dkms/evdi/1.14.7/build/make.log for more information.
Deleting module evdi-1.14.7 completely from the DKMS tree.
ERROR: Failed to install evdi to the kernel tree.
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory

Terminal output

mustekito, 2025-02-19:
Hi,

For me the solution was this one: https://www.displaylink.org/forum/showthread.php?t=69412

Please note that you must install version 6.0, not the latest.

Regards

nomevini, 2025-02-19:
Hi,

For me, the solution was to install the following versions:

DisplayLink: [Version 6.0](https://www.displaylink.org/forum/showthread.php?t=69412) (not the latest)
Silicon Motion: [Version 2.21.2.0](https://oemdrivers.com/graphics-silicon-motion-sm76x-linux-2-21-2-0#google_vignette)

Make sure you install these specific versions, as newer ones may not work properly.

closed 2025-02-19 by nomevini