AdnanHodzic/displaylink-debian #990: Debian 12 black screen, formerly having flickering cursor

SynthGal, 2025-05-02:
```
OS: Debian GNU/Linux 12 (bookworm) x86_64 
Host: Laptop AB 
Kernel: 6.1.0-32-amd64 
Uptime: 8 mins 
Packages: 4003 (dpkg), 17 (flatpak) 
Shell: bash 5.2.15 
Resolution: 2256x1504 
DE: Xfce 4.18 
WM: Xfwm4 
WM Theme: Default-xhdpi 
Theme: Adwaita-dark [GTK2/3] 
Icons: HighContrast [GTK2/3] 
Terminal: guake 
CPU: 11th Gen Intel i7-1165G7 (8) @ 4.700GHz 
GPU: Intel TigerLake-LP GT2 [Iris Xe Graphics] 
Memory: 3528MiB / 15777MiB 
```

Running on a Framework 13 laptop with above specs.

Had the flickering cursor issue, `--debug` output was as follows (note the lack of "up and running" under Displaylink service status, even though I was able to use the software to extend the display)

```
-------------------------------------------------------------------
Starting Debug ...
Did you read Post Installation Guide? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/post-install-guide.md [y/N] y
Did you read Troubleshooting most common issues? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/common-issues.md [y/N] y
--------------- Linux system info ----------------
Distro: Debian
Release: bookworm
Kernel: 6.1.0-32-amd64
---------------- DisplayLink info ----------------
Driver version: 1.14.9
DisplayLink service status: 
EVDI service version: 1.14.9
------------------ Graphics card -----------------
Vendor: i915
Subsystem: TigerLake-LP
VGA: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 01)
VGA (3D): 
X11 version: 21.1.7-3+deb12u9
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf
-------------- DisplayLink xorg.conf -------------
File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
-------------------- Monitors --------------------
Providers: number : 2
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 5 outputs: 6 associated providers: 1 name:Intel
Provider 1: id: 0x56f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
-------------------------------------------------------------------
Use this information when submitting an issue (http://bit.ly/2GLDlpY)
-------------------------------------------------------------------
```

Ran through the instructions as per https://github.com/AdnanHodzic/displaylink-debian/issues/192#issuecomment-452235074 (with sudo to prevent "permission denied" errors)  and now all I get is a black second monitor. Now `--debug` is as follows

```
-------------------------------------------------------------------
Starting Debug ...
Did you read Post Installation Guide? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/post-install-guide.md [y/N] y
Did you read Troubleshooting most common issues? https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/common-issues.md [y/N] y
--------------- Linux system info ----------------
Distro: Debian
Release: bookworm
Kernel: 6.1.0-32-amd64
---------------- DisplayLink info ----------------
Driver version: 1.14.9
DisplayLink service status: 
EVDI service version: 1.14.9
------------------ Graphics card -----------------
Vendor: i915
Subsystem: TigerLake-LP
VGA: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 01)
VGA (3D): 
X11 version: 21.1.7-3+deb12u9
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf
-------------- DisplayLink xorg.conf -------------
File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
-------------------- Monitors --------------------
Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 5 outputs: 6 associated providers: 0 name:Intel
-------------------------------------------------------------------
Use this information when submitting an issue (http://bit.ly/2GLDlpY)
-------------------------------------------------------------------
```

I've hit the point where I am no longer comfortable debugging on my own because my previous attempt has made the problem worse. If you have the time, I could use some help.

gnutechie, 2025-08-04:
Change this:

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

to this:

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "modesetting"
EndSection

And restart lightdm and hopefully this works out for you.

hpsaturn, 2026-08-24:
I had the same error, but I fixed it doing the next config for my Debian 12:

Change the xorg config:

```conf
Section "Device"
  Identifier "AMD_Graphics"
  Driver     "amdgpu"
EndSection

Section "Device"
  Identifier "DisplayLink_Virtual"
  Driver     "modesetting"
  Option     "PageFlip" "false"
EndSection
```

For Sunshine:

`Configuration->Advanced->Force a Specific Capture Method: X11`

My config and setup:

```conf
--------------- Linux system info ----------------

Distro: Debian
Release: bookworm
Kernel: 6.1.0-52-amd64
Driver version: 1.15.0
DisplayLink service status: up and running
EVDI service version: 1.15.0
Vendor: amdgpu
Subsystem: Inc.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev c8)
VGA (3D): 
X11 version: 21.1.7-3+deb12u13
X11 configs: /etc/X11/xorg.conf.d/20-displaylink-amd.conf

-------------------- Monitors --------------------
Providers: number : 5
Provider 0: id: 0x55 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 3 associated providers: 4 name:AMD Radeon Graphics @ pci:0000:07:00.0
Provider 1: id: 0x130 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xff cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xce cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x9d cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```