AdnanHodzic/displaylink-debian #991: Operation Not permitted

idanceinsbs, 2025-05-07:
xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x5b cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 9 associated providers: 0 name:AMD Radeon 780M @ pci:0000:c4:00.0

post deployment and troubleshooting directions followed

idanceinsbs, 2025-05-07:
--------------- Linux system info ----------------

Distro: Ubuntu
Release: plucky
Kernel: 6.14.0-15-generic

---------------- DisplayLink info ----------------

Driver version: 1.14.9
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Inc.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Phoenix1 (rev c1)
VGA (3D): 
X11 version: 21.1.16-1ubuntu1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x5b cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 9 associated providers: 0 name:AMD Radeon 780M @ pci:0000:c4:00.0

idanceinsbs, 2025-05-07:
[make.log](https://github.com/user-attachments/files/20074698/make.log)

DrAzraelTod, 2025-08-12:
had a similar problem, was solved by starting the script not from root, but from the currently logged in user, using sudo

(so using 'sudo su -'  and typing a whole bunch of things is a bad idea here)

might be similar if session is capsuled into tmux or other stuff?